gilestrolab/ethoscope #177: Ethoscope cannot connect to mariadb server with new connector update (8.0.30)

ggilestro, 2022-10-12:
New images (2022) sport a new version of python-mysql-connector (8.0.30). This means the syntax to connect to mariadb databases is now different (as per [here](https://stackoverflow.com/questions/73244027/character-set-utf8-unsupported-in-python-mysql-connector)). This makes the SQLwriter fail miserably.

pepelisu, 2022-10-17:
commit 3475e8ee485f5b6892aaff4a752661b4a0ba7cf0 uses import from ethoscope src.
In my installation, the module ethoscope is not instalelled by default in the node. That means that in a normal installed node, the ethoscope module does not exist and the importing is falling:
`
Oct 17 10:12:33 node python[289212]:   File "/opt/ethoscope-node/node_src/ethoscope_node/utils/mysql_backup.py", line 9, in <module>
Oct 17 10:12:33 node python[289212]:     from ethoscope.utils.io import SQL_CHARSET
Oct 17 10:12:33 node python[289212]: ModuleNotFoundError: No module named 'ethoscope'
`

ggilestro, 2022-10-17:
Sorry about that. I believe (I hope) that in the default node installation (the one that comes with the liveCD) the ethoscope libraries are properly installed. If not, this can be done easily with the following command:

```
cd /opt/ethoscope-node/src
sudo python setup.py develop
```

closed 2023-06-29 by ggilestro