gilestrolab/ethoscope #180: request: sleep depriver module instructions/arduino code

rjs360, 2023-06-14:
Hi all, I am looking to build a few sleep depriver modules for my ethoscopes.  

A few years ago there was fairly detailed documentation on building and running (via arduino code) the dc motor based "tube spinner" module (referred to as the "rotational module" here: https://www.science.org/doi/10.1126/sciadv.aau9253), but (I might be missing something) I can no longer find this information.  

Might anyone still have information/files/instructions for these?  Thanks for any info on this.
-Ruud

ggilestro, 2023-06-15:
Hi. The ethoscope manual still lives at [http://lab.gilest.ro/ethoscope ](http://lab.gilest.ro/ethoscope) 
Instructions for the old module can be found [here](https://www.notion.so/giorgiogilestro/Optomotor-Module-2d7e9f5aa0e84a3d8bbbff62c7014f24). 
Instructions to the new module can be found [here](https://www.notion.so/giorgiogilestro/New-modular-structure-and-new-electronics-a478fcbff07c469db1b531a7f50d336d)

closed 2023-06-15 by ggilestro

pepelisu, 2023-06-15:
> Hi. The ethoscope manual still lives at [http://lab.gilest.ro/ethoscope ](http://lab.gilest.ro/ethoscope) Instructions for the old module can be found [...]

Just as note. The links are asking for a login into notion.

ggilestro, 2023-06-15:
@pepelisu this is worrying. are you sure about that??? They should be openly shared to the world and If I open those in incognito I am not asked for a notion login.