gilestrolab/ethoscope #183: async database writer has stopped unexpectedly

FENGZHENGHFUT, 2023-09-12:
When I run ethoscope, it often got the following error after 8-9 hours and then stopped.
[image]
This maked it hard to continue my experiments, because when I restarted the ethoscope and restarted tracking,  the data will save the new .db file but has the same date. Like  2023-09-11_01-35 and  2023-09-11_09-35. so i don‘ know how to write my metadata.csv's 'date'.
 I'd appreciate it if you have any ideas on how to solve it !

ggilestro, 2023-09-12:
This is normally a sign of a faulty or low quality SD card.

FENGZHENGHFUT, 2023-12-28:
Hi，ggilestro
    I had replaced the different SD cards and it was stable for 3 months. As shown in the image below：
[image]
   I've recently encountered a new problem. I burned and replaced the SD card with a new one, and the first few times my node found the corresponding ethocope number normally, but after that the node couldn't find the ethoscope after powering up the device.  However,my router showed the ethoscope device  connected to the wifi.   Is it the  problem of quality of the SD card? Because when I burn a new card again and load the device the node can find the new ethoscope 000. Would there be any other factors that would cause this, as it would be too expensive to replace the SD card frequently? 
   The node log shows the following：
[image]
   This situation did not affect my node to find other ethoscope devices.
   Looking forward to for your reply.
Thanks you.

closed 2025-06-29 by ggilestro