gilestrolab/ethoscope #184: Resource Tab issue

Dulfy02, 2024-01-23:
I am new to linux but I managed to set up a virtualbox to run the node. Everything seems to be running fine for now, but the one thing that I cant do is access the Resource Tab. I thought it might be an internet problem but pinging google.com gives me no problems. Im not sure if I didnt configure something correctly, but I can do a node update successfully as far as I can tell. Any insight would be appreciated :)

ggilestro, 2024-01-27:
Hi there, 

I have moved all my servers to a more powerful machine and forgot to move the ethoscope resource server. My bad!

It is  now done and it's working properly - in fact even better than before because [I've changed the URL to use the default port which is no longer intercepted by firewalls](https://github.com/gilestrolab/ethoscope/commit/4768bc0b436c10c0d54800dfa406e47205e4191b). Please update your node and this should work now as expected.

closed 2024-01-27 by ggilestro

Dulfy02, 2024-01-29:
Hello, I updated node and still seems that I cant access the resource tab. I have version [dev]61629f...(2024-01-27 15:25:50). When I got to resources it just says wait a few seconds, your node must be connected to the internet for this page to load. I will try just starting over with a new node and see if that works. Thanks for the previous quick reply.

ggilestro, 2024-01-29:
Try with a hard-refresh: shift + F5

Dulfy02, 2024-01-29:
I pulled up the debugger and i get Cross-Origin Request Blocked: the same origin policy disallows reading the remote resource at https://lab.gilest.ro:8088/resources. (reason: CORS request did not succeed). Ill try to look into it.

ggilestro, 2024-01-29:
The node should be trying to pull https://ethoscope-resources.lab.gilest.ro/resources and /news. They both return JSON

https://ethoscope-resources.lab.gilest.ro/ should return a human readable page.

Dulfy02, 2024-01-29:
[image]

Dulfy02, 2024-01-29:
This is what I am getting. Seems like it is trying to get a signal every time I refresh, but its not going though.

ggilestro, 2024-01-30:
It's trying to connect to the wrong URL. The new live is the one I posted above. Either the update was not successful or you still have the old JavaScript in cache. Try a hard refresh

Dulfy02, 2024-01-30:
Hard refresh worked! Thanks for the help.

ggilestro, 2024-01-31:
Glad to hear. 

Unrelated: I just made a dockerised version of the node. Should be more convenient to use than the virtualbox image. https://hub.docker.com/repository/docker/ggilestro/ethoscope-node/general