gilestrolab/ethoscope #187: Ethoscope settings not working

SofiPol, 2024-03-14:
Hi, I'm having a problem setting up my ethoscope. Once connected it appears as ETHOSCOPE_000 and when I try to change the name in "settings" I don't see any list...only update or cancel. Any idea why?

Cheers,

ggilestro, 2024-03-19:
Hi Sofia, I cannot reproduce this problem so I would need to have more information to understand what the problem is.
Which image file did you use and are you running the most recent version of the platform?

SofiPol, 2024-03-20:
Hi Giorgio,

I downloaded the last version of the image (found in Notion) and the same for Manjaro. Could it be a problem of the network? I'm using a laptop connected to the router with a USB adaptor because I don't have an ethernet port, I'm not sure if this could make the Wi-Fi connection more unstable. I'm going to try now with a desktop computer with ethernet port to see If it works better.

Thanks!

closed 2025-02-18 by ggilestro