gilestrolab/ethoscope #188: Ethoscope stops working randomly

BrainFriedLab, 2024-04-04:
Hi there,

I'm certainly a novice at all of this, but we've had a great ton of success troubleshooting. But now, I'm hitting a wall. I can't figure out why this is happening. The ethoscope just stops tracking at random times throughout the run. We were using an ethoscope that had the 2020 image on the micro sd card. It worked well, but then it crashed this week.

I burned the most recent image onto a sandisk 64 GB sd card (only brand I could get quickly to get the experiment back up and running). This worked as expected, but I started a run on Tuesday and came back wednesday to find out it stopped (only collecting about 18 hours of data). I then started another run last night and came back this morning to find that it only collected data for about one hour.

When I click the logs, this is what I see. The error logs here are not from the time it crashed, but instead seem to be right when I click the "view logs" button.

[image]

Maybe I have a setting screwed up or something for this 2023 image?

ggilestro, 2024-04-05:
Hi there!

The errors you are seeing in your logs just refer to the fact that the ethoscope suddenly goes offline. 
Sudden crashes like this one at variable times are normally due to a hardware problem. The most common culprit is the SD card, the second most common culprit is insufficient power from the ethoscope power supply but a worn out ethoscope is not to be excluded. 

Regarding the data backup folder: if the data backup folder is not correctly configured the node may fail but the ethoscope should not. If you see that your data are not saved correctly, I just recommend using no blank spaces in the folder path. In your case, you should rename the mount point from "Memorex USB" to "Memorex_USB" or something like that. This implies changing the actual folder name, not just the configuration. 

Hope this helps.

BrainFriedLab, 2024-04-05:
I just got back to the lab this morning and it's still running! I guess just updating it helped do the trick. I have some back up SD cards, however, in case this happens again. And I guess I should make "building an ethoscope" a class activity. We're a lab of only undergrads here at Rutgers Camden, just outside Philly. This has been such a great system to help teach them troubleshooting, independence, and research methods. THANK YOU for building and supporting it!!!

I'll make the changes to the folder as recommended. I appreciate your help!

closed 2024-11-19 by ggilestro