gilestrolab/ethoscope #189: Database File structure incorrect?

BrainFriedLab, 2024-04-11:
Hi there,

I'm so close to getting rethomics working, I can taste it. BUT, I've found a weird problem with the .db files that is holding me up. Below is a comparison between the database file structures of my .db file (left) and your rethomics ethoscope .db file (right). The csv table is aokay looking. But it's missing tables for all of the ROIs and missing "Start events" and "Var_map" tables which is preventing me from loading up the data into rethomics.

[image]

BrainFriedLab, 2024-04-11:
Figured it out! I went back to more .db files. Turns out my other .db files are a-okay in terms of the tables that are in them. It's just this single one that doesn't have the tables correctly in there. Not sure how to add proper tables, but I've gotten through that error!

ggilestro, 2024-04-14:
Good to hear!

closed 2024-04-14 by ggilestro