gilestrolab/ethoscope #190: ROI targets not detected under infrared lighting conditions

augustinhebert, 2024-05-03:
Hi, 

I am a veterinary student in Nantes, France, and currently working with ethoscopes for my vet thesis, within a team of vets/INRAE scientists (National Research Institute for Agriculture, Food and the Environment).  

We struggle with light reflections on our glass tubes, despite the use of plastic panels that were supposed to diffuse the light. 
To try and prevent this, we decided to use an infrared pass filter (LEE FILTERS 87c) that blocks visible light, in front of our cameras, so that we record only the infrared lighting from the lightbox, whether it is dark or light conditions. 

Our problem is that the black dots we use as targets (using the pdf provided by Q. Geissmann) seem to not be detected under these conditions (we had no problem tracking without the filter). The debug images show that we can quite easily distinguish the targets, but they seem to not be "sharp" enough. 
Our targets are made of paper and sticked on a polyethylene plate set between the lightbox and the tubes, that does a good job at creating an homogenous infrared backlighting. 

I tried :
- setting the focus of the camera manually on a wide range of positions
- putting the filter in different locations : directly sticked on the lens, or on a mount a few millimeters from the lens, and i even tried gently placing a tiny disc of filter between the lens and the sensor of the camera.
- changing the size of the targets
- lining the targets with discs of aluminium foil to increase contrast as the aluminium partially blocks infrared
- increasing contrast with a "brighter" infrared lighting, using the IR LEDs from a second ethoscope. 
None of this worked, but the debug images showed a clear improvement of sharpness and contrast with these measures. 

[image]
[image]
[image]
[image]

The tracking sometimes succeeds but the ROIs are then completely random, showing that the targets are not correctly detected. 

Has someone already encountered this issue ? 

I'm thinking of recording a video, then virtually adding black dots on the correct spots as a post-treatment, and reusing this edited video as the input for an offline tracking. Is that a possibility and if so, how would you do this ? 

Would there be any other solution ? 

Thank you in advance,

Augustin Hebert

ggilestro, 2024-05-06:
Hi Augustin, 

the easiest solution would be to make the targets more visible, either increasing illumination or using a paint/material that is more IR-reflective so that the camera could see the dots with greater contrast. Personally I don't have experience with this but I am positive there should be a straight forward solution. Perhaps cut to size some bicycle-reflective stickers?
 Alternatively, you could start the tracking without the filter and position the filter only at a second time?

augustinhebert, 2024-05-07:
Hi @ggilestro, thank you very much for your answer ! 

I have tried to increase contrast using aluminium foil and it was not sufficient, but the reflective stickers seem like a great idea. i will try that next week and keep you updated. 

I considered this other alternative but I was afraid that setting the filter after starting would slightly move the camera and potentially disturb tracking. But if I don't manage to start tracking with the filter on, I'll end up doing that (I thought about custom-making a plastic mount for the filter, that would fit on the part that holds the camera).

ggilestro, 2024-05-07:
Please let me know if you make any progress. There may possibly be some software solutions too but they would require coding and testing and are therefore not immediate

augustinhebert, 2024-05-29:
Hi @ggilestro,
The solution that worked for me was indeed to start tracking without the filter on (in light conditions) and mounting the filter afterwards. I noticed I sometimes have an tracking error when mounting the filter (something about float division by zero), but this error is less frequent when i mount the filter quickly and spend less time messing around in the field of view of the camera. 
If I am quick and careful, the tracking continues fine under infrared only condition, without a spatial shift of the ROIs.

Thanks !

closed 2024-05-29 by ggilestro