gilestrolab/ethoscope #191: Video_backup service not working inside docker

Anuragbio, 2024-05-17:
Dear Dr. Gilestro, 

We are using the Ethoscope platform for the behavioral monitoring of flies. We have successfully tracked flies using the track option on the node, and the backup is stored at /mnt/ethoscope_data/results/0011837902f74aaf86c865e45a1126ca/ETHOSCOPE_001.

However, when we click the record video button, the recording starts, and the video is saved on the Ethoscope device. Unfortunately, no video can be found on the node. We have looked into the available solution here (https://github.com/gilestrolab/ethoscope/issues/146), but since we are using Docker, we cannot run systemctl commands.

Please let us know how we can troubleshoot or what information can help you to rectify this issue.

Ethoscope device:

Hardware info
    • SD card age: 11 days, 3h, 50min, 5s 
    • SD card version: 20230306_ethoscope_000.img 
    • PI version : Raspberry Pi 3 Model B Plus Rev 1.4 
    • PI camera : {"IFD0.Model":"RP_imx219","IFD0.Make":"RaspberryPi","version":"PINoIR 2"} 
Ethoscope node:
Platform- Docker

Version: [master] b73444... (2024-01-27 11:30:45)

Best regards,
Anurag

ggilestro, 2024-05-17:
Assuming you mounted your volumes using the default destinations in the `docker-compose.yml` file, the videos should be in the folder `/mnt/ethoscope_data/videos/`. They are copied there from the service called `ethoscope_video_backup` which should be running by default even with the docker container. 

Keep in mind the Docker solution is still rather experimental. If you can, I'd advise using a dedicated installation.
Logging with the docker can be accessed using the docker command: `docker logs ethoscope-node`. If the service runs normally, you should see something like:

```
ethoscope_video_backup.service: INFO:root:Starting backup round
ethoscope_video_backup.service: INFO:root:Found 0 devices online: 
ethoscope_video_backup.service: INFO:root:Backup finished at t=1715940387
ethoscope_video_backup.service: 
```

ggilestro, 2024-05-22:
Judging from these lines:

```
ethoscope-node | ethoscope_video_backup.service: INFO:root:Initiating backup for device 0011837902f74aaf86c865e45a1126ca
ethoscope-node | ethoscope_video_backup.service: INFO:root:Backup done for for device 0011837902f74aaf86c865e45a1126ca
```
The video seems to backup just fine. Are you sure you are looking in the right location?

Anuragbio, 2024-05-22:
Thank you for your reply. We are using the YML file below in which "/mnt/ethoscope_data" is the folder where we can find .db files (complete folder path "/mnt/ethoscope_data/results/0011837902f74aaf86c865e45a1126ca/ETHOSCOPE_001"). However, in the case of video files, they can only be found on ethoscope and not in the video folder of this path "/mnt/ethoscope_data/".

## Contents of YML file
"version: '3.8'

services:
  ethoscope-node:
    build: .
    image: ggilestro/ethoscope-node
    container_name: ethoscope-node
    restart: always
    volumes:
      - /mnt/ethoscope_data:/ethoscope_data
      - ./etc-ethoscope:/etc/ethoscope
    network_mode: host

  vsftpd:
    image: metabrainz/docker-anon-ftp
    container_name: ethoscope-vsftpd
    depends_on:
      - ethoscope-node
    restart: unless-stopped
    network_mode: host
    environment:
      MAX_CLIENTS: "3"
      MAX_PER_IP: "2"
      FTPD_BANNER: "Welcome to the public ethoscope data FTP server of our laboratory"
    volumes:
      - /mnt/ethoscope_data:/var/ftp:to"

ggilestro, 2024-06-09:
I am sorry but I cannot reproduce this. Video backup works out of the box for me on dockers, both when the containers run in linux or windows (I have not tried macOS).

closed 2024-11-19 by ggilestro