gilestrolab/ethoscope #192: Video conversion for offline tracking not working

augustinhebert, 2024-05-29:
Hello, 

I am trying to perform offline tracking on my ethoscopes but I struggle with video conversion

After flagging my ethoscope as experimental, I click on **Convert Video** on the UI. I then click on **Start Tracking** and the videos are quickly available in the list of "MovieVirtualCamera". 
My problem is that when I try to record a new video, I convert them afterwards and the new video doesn't show up in the list, no matter how long I wait (several hours for minute-long videos). 

When I ssh into my ethoscope, I can locate the newly recorded h264 chunks in their correct folder (/ethoscope_data/results/identification_of_the_ethoscope/ETHOSCOPE_XXX/YYYY-MM-DD_HH-MM-SS), but not the corresponding concatenated mp4 file. (For the videos recorded before flagging as experimental, I have the h264's and the mp4 file)

When I look in the logs of the ethoscope on the UI just after clicking on **Convert**, it says `succesfully processed files in folder /ethoscope_data/results/identification_of_the_ethoscope/ETHOSCOPE_XXX/YYYY-MM-DD_HH-MM-SS` but i cannot see the processed file. 

I tried `sudo python  /opt/ethoscope-device/scripts/tools/process_all_h264.py` but it again tells me everything has been successfully processed but i see nothing. 

I thought my SD card lacked available space to perform conversion so I deleted the mp4 files that i got from the previous recordings, leaving me with only h264's (from before and after flagging as experimental). I then retried converting but the only h264 processed are those recorded beform flagging, so I get the same mp4 files as before, but still nothing corresponding to my new recordings.

I tried de-flagging as experimental, rebooting, tried to record several times but all with the same result

If anybody could help that would be great, I hope I was not too unclear ! 

Thanks in advance

Augustin Hebert

ggilestro, 2024-07-15:
Hi Augustin - sorry for the delay with my reply. 
Can you please describe more clearly what it is that you are trying to achieve?

If you want to merge multipled h264 videos into one big mp4, then you should do the following:
1. backup the video contents from your ethoscope to the node. This should be done automatically if the `ethoscope_video_backup` service is running
2. *on the node*, run the conversion script. A newer version of it is located in `accessories/h264_to_mp4.py`. If you run this with the --help flag it will tell you how to use it

```
Usage: h264_to_mp4.py [options]

Options:
  -h, --help            show this help message and exit
  -p PATH, --path=PATH  The root path containing the videos to process
  -l, --list            Returns a list of folders containing mp4 files
  -e EXTENSION, --extension=EXTENSION
                        The extension of the video file chunks generated by
                        the ethoscope
  --force               Force recreating videos even when MP4s are present
```

augustinhebert, 2024-07-22:
Hi Giorgio, 

Thank you for your answer, 

I was indeed trying to merge my h264 files into one mp4 file, following the instructions for offline tracking on the Notion page, but these didn't work for me. (My goal was to test several modifications on the adaptive_bg_tracker.py file with the same video input in order to compare how those modifications affected tracking)

In the meantime I found a workaround : I downloaded my h264 files from the node on my computer with FileZilla, I then used ffmpeg to merge and convert them into one big mp4 file, and I transferred this mp4 file back on the ethoscope for offline tracking, which worked. Not very efficient but I managed to do some testing that way !

Thanks for the instructions, sadly I am currently not able to work on my ethoscopes as I cannot access the node due to network issues at my school... Unless there are commands I can run on the ethoscopes to make them work without the node ?

I will try it as soon as possible ! 

Augustin Hebert

ggilestro, 2024-07-22:
I see. Offline tracking is kind of an experimental feature and may not work 100%. 
You can start/stop experiments on the ethoscope without the use of the node using the ethoclient command but it's not documented at all and you'll have to figure out from the [code](https://github.com/gilestrolab/ethoscope/blob/dev/src/scripts/ethoclient.py), sorry.

closed 2024-07-22 by ggilestro