gilestrolab/ethoscope #198: Node (LocalHost) suddenly stopped working

Sid0li, 2024-12-05:
Hello, i was about to load an experiment and when i opened the local host to connect the ethoscopes the local host appeared brand new with no memory of ever detecting any device. I went to the "update system" section and restarted and updated the node and its been loading for like an hour as seen in the image below

[image]
 
And now when i try to update/restart the node this appears

[image]

Ive been using this node for like 6 months already and the experiments section appears empty too (it was like 2/3 pages long)

What can i do?  ideally without reinstalling the whole thing again :p

Sid0li, 2024-12-05:
1. I tried and restarting it changed nothing unfortunately
2. Nop, the Node is based on a computer
And the disk is at 37% capacity so it has plenty of space. Maybe it failed but nothing else in the computer seems damaged :c

ggilestro, 2024-12-06:
I think the easiest fix would be to run the script below on your node as sudo. It will delete the current version of the software and download the most recent one. You will start from scratch without having to reinstall the whole thing.

```
#!/bin/bash

# Define variables
REMOTE_REPO_URL="https://github.com/gilestrolab/ethoscope.git"
LOCAL_REPO_URL="git://node/ethoscope.git"
CLONE_DIR="/opt/ethoscope-node"
BARE_REPO_DIR="/srv/git/ethoscope.git"
BRANCH="dev"
USE_REMOTE=false

# Function to display usage
usage() {
    echo "Usage: $0 [--remote | --local]"
    exit 1
}

# Check if the script is running with sudo privileges
if [ "$(id -u)" -ne 0 ]; then
    echo "This script must be run as root or with sudo privileges."
    exit 1
fi

# Parse arguments
while [ "$#" -gt 0 ]; do
    case "$1" in
        --remote)
            USE_REMOTE=true
            ;;
        --local)
            USE_REMOTE=false
            ;;
        *)
            usage
            ;;
    esac
    shift
done

# Determine the repository URL and fetch if local
if [ "$USE_REMOTE" = true ]; then
    REPO_URL="$REMOTE_REPO_URL"
    echo "Using remote repository: $REPO_URL"
else
    REPO_URL="$LOCAL_REPO_URL"
    echo "Using local repository: $REPO_URL"

    # Run git fetch to sync the bare repo, assuming it should be done as the git user
    sudo -u git git -C "$BARE_REPO_DIR" fetch origin || { echo "Failed to fetch from local bare repository. Exiting."; exit 1; }
fi

# Change directory to /opt
cd /opt || { echo "Failed to navigate to /opt. Exiting."; exit 1; }

# Remove the existing directory if it exists
if [ -d "$CLONE_DIR" ]; then
    echo "Removing existing directory $CLONE_DIR"
    rm -rf "$CLONE_DIR" || { echo "Failed to remove directory. Exiting."; exit 1; }
fi

# Clone the appropriate repository
echo "Cloning repository $REPO_URL"
git clone "$REPO_URL" "$CLONE_DIR" || { echo "Git clone failed. Exiting."; exit 1; }

# Change directory to the cloned repository
cd "$CLONE_DIR" || { echo "Failed to navigate to $CLONE_DIR. Exiting."; exit 1; }

# Checkout the specified branch
echo "Checking out branch $BRANCH"
git checkout "$BRANCH" || { echo "Git checkout failed. Exiting."; exit 1; }

# Set origin to local
git remote set-url origin $LOCAL_REPO_URL

echo "Repository setup complete."
```

Sid0li, 2024-12-10:
Hello Giorgio! We tried running the script you sent us but the node still didn't work.... Before asking you, we had tried (and probably altered) a few things so maybe that's why it didn't work. Still we decided we want to reinstall the whole Manjaro operating system and the node because the PC was not really working well from the beginning, the PC froze quite often.

When we first installed it, we used Option 2 from the Ethoscope User Manual (Installing the node using the package manager) because we weren't able to install it with Option 1 (Installing the node using the liveCD). Now that we're trying again, we want to use Option 1, which is the recommended option according to the manual. We reboot the PC with the USB pendrive with the image of the Manjaro+ethoscope node and we boot that image. The Manjaro installation page appears and we boot it. But then it never finishes booting, instead the computer is stuck in this step:
[image]
This is the same issue we had when we first tried this installation.

Do you know how we can go on with the installation? Or do you recommend any other way to reinstall the whole thing?

Thank you very much for your help

ggilestro, 2024-12-10:
Hi - The liveCD is a bit old and needs to be updated. You are better off using option 2 for now.

ggilestro, 2025-02-18:
The liveCD has been updated to the newest Manjaro Version.
The latest version, dated Feb 14 2025, can be downloaded from [here](https://imperialcollegelondon.box.com/shared/static/ha14uo1c634u5qpxvfvksh8d3y0vpnjt.iso) and it's linked in the ethoscope manual too.

closed 2025-02-18 by ggilestro