gilestrolab/ethoscope #202: Trouble installing via Docker

MartynasArba, 2025-02-11:
Hi all,

After Docker installation, the node does not run and instead I get this message:
```
ethoscope-node    | Traceback (most recent call last):
ethoscope-node    |   File "/usr/bin/systemctl.py", line 6789, in <module>
ethoscope-node    |     FORCE_IPV4 = opt.force_ipv4                                                                                                                                                        
ethoscope-node    |                  ^^^^^^^^^^^^^^                                                                                                                                                        
ethoscope-node    | AttributeError: 'Values' object has no attribute 'force_ipv4'
```
Is there a solution, guide, or any other steps I should take to complete the installation successfully?

Steps I took leading to the issue:
1. Cloned the Docker folder from github (I've also tried using `docker pull`, which led to issue #185 )
2. Navigated to the Dockerfile and installed it using either using `docker build .` or `docker compose up`
3. Ran it, resulting in the error

Any help would be greatly appreciated!

MartynasArba, 2025-02-12:
Thank you for the reply!
Docker is running on Windows 11.

ggilestro, 2025-02-12:
I have restructured and updated the Docker stack. You can get the new files on the r[epo (make sure you use the dev branch)](https://github.com/gilestrolab/ethoscope/tree/dev/Docker/node) or the image on [dockerhub](https://hub.docker.com/repository/docker/ggilestro/ethoscope-node/general) 

I have not tested it in Windows11 yet. Perhaps you can give it a try and let me know!

MartynasArba, 2025-02-14:
Both options (`docker pull` using the image and `docker build` using the Dockerfile from the dev branch) now result in seemingly nothing - it runs for 0 seconds and exits with status 0, creating a single container with a name specified or generated when building, which behaves the same way if trying to run it. 

I also tried using `docker compose up` for the dev branch files. This creates the following container structure:

[image]

From here, the `node-base-1` keeps restarting with a message `node-base-1 exited with code 0`, but the main `node` container keeps running, with various (seemingly correct?) outputs from these containers:
- `ethoscope-node`
- `ethoscope-git-server`
- `ethoscope-node-backup`
- `ethoscope-vsftpd`
- `ethoscope-node-video-backup`
- `ethoscope-node-update`

My guess is then that something is wrong with `node-base-1`, but after playing around for a bit I still can't figure it out. I'll try setting up Linux on a spare laptop in the meanwhile. 

Let me know if there's anything else to check or to try, as it'd be really nice to run the node on my main workstation!

MartynasArba, 2025-02-14:
It seems the issue is that `network_mode: host` [only works on Linux](https://docs.docker.com/engine/network/drivers/host/). 
I'm trying to find a workaround for Windows or Mac, and I'll post an update here if I find a solution. 
Thanks for all the help!

ggilestro, 2025-02-14:
I see. Thanks for looking into it. 

One can do automatic port binding and overcome the use of `network_mode: host` but the ethoscope-node container needs to access port 5353 (at least) for the avahi-bonjour-mDNS automatic discovery of ethoscopes on the network and that will most likely crash with Windows11 native service.

I think the easiest solution would be to abandon windows and find a computer to install linux onto. The current installation image is a bit old but if you use an archlinux based image (e.g. Manjaro) you can get the node running simply with `yay -S ethoscope-node` (or use the dockers)

ggilestro, 2025-02-14:
I have made [a new ISO image](https://www.youtube.com/watch?v=S7Rl6XgkWUg). This is once again the easiest way to setup the node. I have added some notes to the docker-compose file to explain what the problem is but I am afraid not much more can be done to solve the Docker issue.

closed 2025-02-14 by ggilestro